webhdx/PicoBoot #91: Original Ipl don't start

Fgo988, 2023-06-23:
| Answer some general questions first                             	  |  	     |
|:-------------------------------------------------------------------	|:------:|
| Which firmware version are you using?                             	| `v0.2` |
| Are you using the official Raspberry Pi Pico board?                 | yes |

Hallo,

I installed a few Picoboots before, but with this I have a strange Problem.
When the SD Card is inserted Swiss will start perfectly.

But when I hold the A Button or the SD Card is not inserted the Original IPL will not start and I have to restart the Gamecube. It is stuck at "Rebooting into Original IPL".

When start the Original IPL out of swiss it will start without any problem.

Can somebody help me with this problem?

webhdx, 2023-06-23:
When you get into iplboot (program that loads Swiss from SD card), PicoBoot has nothing to do with what happens next. I don't have any idea what could be the issue here. What kind of external devices are you using?

Fgo988, 2023-06-23:
Do you mean with external devices the Controller, TV or SD Card Reader?
I have two original Gamecube Controller. And I already tried it at different TVs. And I use the SD Card Reader from Aliexpress for Gamecubes with Gameboy Player.
I have the same "issue" with all combinations and Controller ports.

webhdx, 2023-06-25:
The problem lies in iplboot since you can load IPL from Swiss. I have no idea what's the problem here. :(

closed 2023-07-24 by webhdx