webhdx/PicoBoot #93: GameCube always boots into normal boot up.

Link267, 2023-07-05:
| Answer some general questions first                             	  |  	     |
|:-------------------------------------------------------------------	|:------:|
| Which firmware version are you using?                             	| v0.2 |
| Are you using the official Raspberry Pi Pico board?                 | yes |

Hello!

I am having trouble booting up into Swiss after the Picoboot install. The GameCube boots into its normal state every time no matter what I’ve tried.

Lights are on the pico board when the console is powered up however I’m not seeing troubleshoot info when holding down the D-Pad. I’m assuming this is due to my smartTV where I have a menu for “Channels” and the HDMIs that are connected but I miss out on the first few seconds of boot up because the tv has to load the input method (takes like 2-3 seconds).

My wires are all below 10cm and I believe the solder job went well even for my first time.

My files are in the correct locations and renamed (I went off the MachoNachoProductions video for install).

I replaced my SD2SP2 adapter with a GC2SD memory card but that didn’t work.

I read that GCHD-MKII also causes issues so I unplugged both the HDMI and the GCHD from the back of the GameCube but it still boots into the normal boot up.

As a test I removed everything to ensure the GameCube is still working fine and it’s perfectly reading disks and playing. After reinstallation it’s the same, reads disks and plays perfectly but never boots up Swiss.

At this point I’ve sunk so much time and money to get what should be an easy mod to even boot up properly. Not sure what the issue is here and I’m tired of tearing down this cube just to check.

[image]
[image]
[image]
[image]

webhdx, 2023-07-05:
Troubleshooting screen will last until you release D-Pad Down button. If you don't see it at all then there is something wrong with soldering. Judging by the photos, it looks all okay but maybe try resoldering both pins on IPL chip. Also try reheating ground wire because it requires much more heat to melt and make good connection.

Link267, 2023-07-05:
Sounds good I’ll solder everything again then. Any recommendations on wire length? I’ve seen multiple answers to this but wasn’t sure what to actually use.

I’m using 26awg wire and the setup in the pics is about 9 and a half centimeters but that leaves literally no room for the pico board.

webhdx, 2023-07-05:
10-12cm is recommended length. The shorter the wires are the better. 26 AWG is okay, I've used it for my installs too but you have to make sure there are no loose strands touching adjacent pins.

Link267, 2023-07-05:
Here is the re-installation job on the motherboard. 

I twisted and tinned the wires beforehand and tried different angles to better see the contacts. 

Can you see anything that stands out before I attach to the pico board?

[image]
[image]
[image]

[image]

webhdx, 2023-07-05:
Everything looks good to my eyes. Just a word of caution - since the wires are quite thick and short you must be very careful to not pull on the cables because it's easy to rip legs from IPL chip.

Link267, 2023-07-05:
It worked! Thank you so much for taking the time to look into this with me, going through threads for over a week I was about to give up. 

Much appreciated and awesome mod!

[image]

closed 2023-07-05 by Link267