webhdx/PicoBoot #94: Can you Build a cheap Bluetooth hub for GameCube

Hi11312, 2023-07-20:
| I'm wondering if you could start developing cheap Bluetooth controller hub for gc  |

| Are you my GitHub sponsor? | no 	|
| Did you check older issues to see if it was requested before? | No |

hoping a this project could be a price range of $10 to 30 
Also Im wondering when you're going to release the m.2 data project for GC and price range?
Thx u for reading
Also sorry I don't know anyway to contact you anywhere else then the issue page

Hi11312, 2023-07-20:
Oops didn't notice the spelling typos 😬

webhdx, 2023-07-24:
Unfortunately this is not what PicoBoot will ever do. PicoBoot is a modchip and it would never try to be anything more than that. Especially Bluetooth controller adapter because it requires to hook into a different bus which would make the installation more difficult.

Please look into https://github.com/darthcloud/BlueRetro. It's open source and has GameCube compatibility. There are tons of different adapters available on the market. 

As for M.2 Loader question - there is no release date nor price announced yet. You have to be patient. To contact me, you can join [my discord server](https://click.webhdx.dev/discord).

closed 2023-07-24 by webhdx