webhdx/PicoBoot #96: was booting into swiss, now only going to vanilla NGC

DecanFrost, 2023-07-26:
| Answer some general questions first                             	  |  	     |
|:-------------------------------------------------------------------	|:------:|
| Which firmware version are you using?                             	| `v0.2` |
| Are you using the official Raspberry Pi Pico board?                 | yes |

So I had installed Picoboot, flashed the board, created a SD card with Swiss, tossed on some ISOs and everything was working great. tested with GC2SD and SP2SD adapter also.

havent used it in a few weeks, went to turn it on today and the Gamecube boots but to Vanillia NGC.  

I held down the D pad for debaug and this is the error msg...

iplboot
Trying sdb
Couldn't mount sdb: A hard error occured in the low level disk I/O layer

not sure why im getting this, i have formated a few more cards with fat32 and exfat  placed swiss and renamed as ipl.dol  and still get this error

any suggestions?

thank ou

webhdx, 2023-07-26:
Do you get the same error on both SD card adapters? This seems to be an issue with SD card setup.

DecanFrost, 2023-07-26:
yes i do, i have tried 4 cards now, different brands too

webhdx, 2023-07-27:
Try different SD card adapter too.

closed 2023-12-31 by webhdx