webhdx/PicoBoot #98: Help for picoboot install - Nothing on screen

gdureuil, 2023-08-10:
| Answer some general questions first                             	  |  	     |
|:-------------------------------------------------------------------	|:------:|
| Which firmware version are you using?                             	| `v0.2` |
| Are you using the official Raspberry Pi Pico board?                 | yes |

Hello There. I'm trying to install picoboot on my DOL-001. Pluto 2 board is installed and confirmed working for HDMI.
When the Pico is soldered, nothing appears on screen, no boot, not even normal boot.

To be noted, the first time I booted, I heard a spark because of bad insulation to heatsink and stopped it immediatly. Desoldered the pico and all boots fine and launching a game, no worries.

Resoldered the pico, and no boot at all. The pico is recognized by computer to reflash the program so it should work. Reduced wire length to 10cm, and still a no go. LED on Pico is on.

[image]

[image]

I also tried a solution in another post, like using 5V on VSYS for pico, and still no boot nothing on screen.

Can somebody please help ? Thank you.

webhdx, 2023-08-10:
Hard to tell. Your soldering looks okay. You should definitely insulate wires above IPL chip. This is a very common mistake with thick wires. But it shouldn't really do any damage. I have no ideas, maybe try another Pico?

gdureuil, 2023-08-10:
Also, when booting with Pico and Pluto2, the Pluto does not stop blinking like it should. Like it's not initializing. Can my problem be linked to power draw ? Is there other points I can test for powering the Pico ?

Thanks for your help.

webhdx, 2023-08-10:
Yes it's also possible. Unfortunately GameCubes are very prone to power problems due to failing capacitors. It's not unusual for 20+ years old console.
You can try a different power supply and/or power regulation board underneath main logic board.

gdureuil, 2023-08-10:
Well, GC powerboards or Pico, I don't have spares right now, so this will wait for a bit. Thanks anyway for your help !

webhdx, 2023-08-10:
If you can access another GC just first check with a different power supply. Then I'd disassemble the console and replace power regulation board (in GameCube, it's under the main logic board):

[image]

If nothing helps then try another Pico.

gdureuil, 2023-08-11:
Ordered a new Pico and will test again soon, that's the easiest part to get for me. Will keep you posted.

gdureuil, 2023-08-12:
Still no go with a new pico. This time it does not prevent the console from booting, but still nothing, and my cables are short.
I also tried powering from VSYS pin from 5V near U11.

Could something have blown on the gamecube that could still make it usable (played 2 hours without pico to confirm) and prevent picoboot from working ?

Also, would it be possible to power the pico from usb (removing solder to gnd and power of course), just to test if it's a power draw problem ?

Thanks.

Photo of new install just in case.

[image]

gdureuil, 2023-08-12:
I think I found out what happened. The short likely killed the LC125A chip near the IPL. If I understand correctly, this is some kind of fuse for serial ports. My chip seems damaged when looking closely with magnifying glasses. Could that be the culprit ?

[image]

webhdx, 2023-08-12:
This is a buffer chip and it indeed can cause PicoBoot to not work. But it's possible to use alternative soldering points to omit this chip. However this is not recommended because you'll need to solder another wire to IPL chip and risk ripping legs off the chip if you are not gentle enough.

Do you at your own risk:

Solder GP5 (pink color on the installation diagram) wire to pin number 1 of the IPL chip. It's the first pin to the right in the top row when looking straight at the motherboard. It's indicated by an arrow.

It should look like this (green wire on the photo):

[image]

With blown U14 chip you'll loose Serial Port 1, so no BBA or M.2 Loader fun for you.

gdureuil, 2023-08-12:
OK that's nice to know. I think I'll go the easy way and order a flex pcb from helder to ease the installation.

If you can, could you also give me the full reference written on this chip ? I'll see if I can find a replacement.

Thank you for you help.

Extrems, 2023-08-12:
I believe this is the part you need.
https://www.digikey.com/en/products/detail/texas-instruments/SN74LVC125APWR/276865

webhdx, 2023-08-12:
Helder's Flex Cable will also work in that case. And it's probably a better solution in your case.

gdureuil, 2023-08-19:
Received my flex cable today and installed everything. What a relief to see swiss booting !
I want to thank your for your help on this install. Lesson learned, if a flex cable exists, go the easy way. And if I ever need M2 Loader, i'll replace the fried chip.

Thanks again kind sir !

closed 2023-08-19 by gdureuil