webhdx/PicoBoot #99: Picoboot bypass in shared space with GCLoader PNP HW2

generalUrsas, 2023-08-11:
| Answer some general questions first                             	  |  	     |
|:-------------------------------------------------------------------	|:------:|
| Which firmware version are you using?                             	| `v0.2` |
| Are you using the official Raspberry Pi Pico board?                 | yes||

I have two succesful installs of Picoboot with SD2SP. I just installed GC Loader PNP HW2 for better data speeds, removed the SD2SP adapter and SDCard, reconfigured and moved SDCard for GC Loader. When I boot, sometimes I get GameCube error screen and sometimes it boots into Swiss from the CGLoader. It seems random. When I get the error screen, there is a longer pause, like some sort of hiccup. When it works, it seems to lauch Swiss almost instantly.

Is there a way to render the PicoBoot inert by "cutting" desoldering a minimal amount of the wires? I'd like to eliminate any potential hardware conflicts for troubleshooting purposes, while leaving the majority of the PicoBoot mod inplace - in case I want to revert to the PicoBoot in the future.

Thank you,
Mike

webhdx, 2023-08-11:
Desolder 3.3V wire from Pico side and insulate it so it doesn't accidentally short. It'll completely disable the modchip and the console will act like a stock GameCube.

generalUrsas, 2023-08-12:
Thank you for the response. It does not seem to be a conflict with the Pico Boot mod, since the problem is now isolated. Great job on the Pico Boot mod. Looking forward to the m.2 product!

closed 2023-08-16 by webhdx