webhdx/PicoBoot #101: picoboot causes waves with component cable

thecoolestever0, 2023-09-11:
| Answer some general questions first                             	  |  	     |
|:-------------------------------------------------------------------	|:------:|
| Which firmware version are you using?                             	| `v0.2` |
| Are you using the official Raspberry Pi Pico board?                 | yes |

[image]
This happens when I use my Retro-Bit Retro Prism Component Cable. Swiss still runs and games load with the same problem. The problem does not occur when I boot into the default gc menu. The analog port works completely fine and if i boot what I want then plug in the digital cable it works. I should also note that it did not go wavy until after detecting the disc until it just started being wavy all the time for digital output (only in pico not default) for no reason.

I have tried changing to 5v in several different locations, changing the power supply and power cord with no change, changing all the pico cables, using a different pico, using a different sd2sp2, and using a different sd card. i have also removed the xenogc as i thought that could have been the problem but it still occurs.

After about two weeks of trying to find a solution i have run out of ideas aside form buying another GameCube and testing to see if that is a problem.

webhdx, 2023-09-11:
Looks like resolution issue. Swiss
May be forcing 480p and your TV can't handle that. Using analog cable change Swiss Video Resolution from Auto to 480i/576i.

thecoolestever0, 2023-09-12:
Thank you changing from auto to 480i fixed the problem.

closed 2023-09-12 by thecoolestever0