webhdx/PicoBoot #102: ipl.h  not found

rdoetjes, 2023-10-02:
Hi,

I checked out PicoBoot but when running cmake and then make all, I get the error ipl.h not found.
And it indeed isn't in the source tree, it is also not a header file of the pico_sdk apparently, so I must be missing something is there an additional git module that I may need to check out?

The reason I want to recompile this, is to relocate the pins so they are in a more convenient location for a Tiny RP2040 and see if e can use those as well.

Thanks!

Ray

webhdx, 2023-10-06:
Hey,

Sorry for late reply. Compilation is described in the Wiki: https://github.com/webhdx/PicoBoot/wiki/Compiling-PicoBoot-firmware

rdoetjes, 2023-10-06:
Awesome thanks! I had not found this wiki.Cool 😎 I’ll get to it.Sent from my iPhone