webhdx/PicoBoot #104: The Pico should be powered on the VSYS pin.

jeromedontdev, 2023-11-04:
I noticed that you are utilizing the 3.3v outpost pin for powering the Pico. For a 3.3v source, the VSYS pin should be used per the datasheet.

webhdx, 2023-11-04:
Yes you are correct however this was done intentionally because I wanted to make the setup as simple as possible without using diodes or MOSFETs. 3.3V instead of 5V was also chosen to minimize the chance of breaking consoles by shorting power to data pins. But I plan to change it in the next release as we found out slow rise time of 3.3V power rail is problematic on some setups.

jeromedontdev, 2023-11-04:
Maybe I am missing something here, but VSYS is tolerant of 1.8v-5.5v and has a built in boost-buck converter, so you can pretty much source power from anywhere. The mosfet is only required if you’re attempting to power on VSYS and plug in USB/VBUS at the same time - an issue that would occur regardless of the pin chosen. I would think using a static voltage power output pin that bypasses all implemented power conditioning would be less simple and more risky.

webhdx, 2023-11-04:
This is all true but actually choosing 3.3V output pin was a conscious decision to not risk people messing with 5 Volts and potentially shorting it to data pins. Trust me we would have a lot of reports of fried CPUs. With current wiring you can connect USB cable and still don't make any harm to the console or Pico. Powering via VSYS with 3.3V and connecting USB cable at the same time would result in feeding 5V to GC's 3.3V power rail. That would end up really bad. Technically speaking you are correct and what we are doing is incorrect but it's actually much safer.

As I've said - I'm now recommending using a diode and feeding power to VSYS:
[image]

jeromedontdev, 2023-11-04:
I see, thank you for elaborating. It’s something that I noticed yesterday and simply hadn’t thought of before after probably 100 installs, so I was just a tad confused.

closed 2023-11-04 by jeromedontdev

thedestroyer80, 2025-02-05:
Ok so if I use a Diode to connect to the vsys to the 5v gc connection would I be able to have the usb cable connected to the pico simultaneously for updates and other whatnots?

webhdx, 2025-02-05:
@thedestroyer80 Yes

thedestroyer80, 2025-02-05:
@webhdx and I don't have to worry about frying my gamecube?