webhdx/PicoBoot #108: Swiss doesn't boot "no dol loaded"

sicounet, 2023-12-31:
| Answer some general questions first                             	  |  	     |
|:-------------------------------------------------------------------	|:------:|
| Which firmware version are you using?                             	| `v0.2` |
| Are you using the official Raspberry Pi Pico board?                 | yes/no |

Hello , i really have no more ideas .  There are not my firsts gamecubes with a picoboot that i do , i never had problems before and i use the exact same things as always . The picoboot is well installed on 3 gamecube , i can do the  d pad down test on all 3 . I used 3 or 4 sandisk ultra 128go , i tried exfat , fat32 , sd2sp2 , gc2sp2 adaptators . i have ipl.dol at the root of my sd card ... i have tried evrything .... It can't be an adaptator issue i've tried all i have on all 3 gamecubes . No matter what i do , in the test screen "no dol loaded" it's like i plugged nothing inside the console. And it can't be the sd cards, either  i use the same since years and i tried 3 sd cards on the 3 different Gamecubes. I've spent hours trying to understand .

webhdx, 2023-12-31:
It's weird. If you are getting into iplboot then PicoBoot is installed fine. Retouching soldering will not fix the issue.

Make sure SD card is formatted with FAT32 and it's using MBR partition table. Also try to do a quick power cycle - it could be an issue with power supply.

sicounet, 2023-12-31:
Yes , as i told you i tried all . fat32 or exfat (exfat was never an issue before) but whatever . I also tried to format one sd card to make it only one partition ( i saw a guy do that) . The power supply i use for all my gamecubes that i mod is an official one . I only have this one and it was fine  . But i want to try your power supply  test but i am not sure to understand what you mean by "do a quick power cycle .

webhdx, 2023-12-31:
Quickly turn off and on the console.

sicounet, 2023-12-31:
mmmm i probably did it without knowing with all the tests i did. But i will give it a try . I ordered a kingstone sd card to ... but i don't believe it's the problem , i never had issues with sandisk before.

sicounet, 2024-01-01:
It changes nothing . What can i say more ... ? One of the gamecube has a xeno gc cheap . But it never was a problem before. My sd cards are sandisk from aliexpress , never had a problem before and it is real sandisk but for the chinese market .  I only need to have an ipl.dol file in the root of my sd ?  It was the swiss_r1400.dol that i changed . I tried with the latest swiss version to . I tried  everything, i am so confused.

webhdx, 2024-01-01:
Is this a problem with just a one console? Have you checked this particular memory card in other GameCubes?

sicounet, 2024-01-01:
yes i have 3 gamecubes motherboards. All 3 with the picoboot well installed . I tried also the 2 adaptators (sd2sp2 and gc2sd on both memory card port)) for the sd on the 3 Gamecubes . What kind of wizardery this is ?! haha

webhdx, 2024-01-01:
But does the card work on any other GameCube?

webhdx, 2024-01-01:
I'd recommend checking out another SD card brand. Chinese SanDisk sounds like a really poor choice to me 🙈 I have a few SanDisks and they work fine but I'd definitely stay away from cards shipped from China. They are often knockoffs.

sicounet, 2024-01-01:
No , i have many sd cards 128go ( all sandisk ). I have try 3 of them. The 3 won't work on the 3 gamecube . I feel i have made a noob mistake so obvious i can't see it. But i did dozens of gamecubes with a picoboot last year and i changed nothing in my way to doing it ... that's why i am soooo lost .

sicounet, 2024-01-01:
For me chinese or not they are sandisks. They work fine on my xstations , fenrir etc...

webhdx, 2024-01-01:
It's not guaranteed to work with GameCube. It uses SPI mode and it's fully controlled via software. Some cards are incompatible. Just check with a different SD card brand - if you get them to work then you have the answer and there is nothing that can be done to make these particular SanDisks to work.

sicounet, 2024-01-01:
yes i ordered a kingstone to try. If nothing change i will try another sandisk not from aliexpress but i don't believe it . It doesn't make sense . If i format one of the 3 sd card i tried with the 3 gamecubes and install it for a xstation or a fenrir it will work . I also try that ...

sicounet, 2024-01-01:
Compatibility says almost all sandisks are compatible. Anyway , thanks for your time , i will post my results with another sd brand later.
Happy new year.

webhdx, 2024-01-01:
Yeah I even own some SanDisks myself and use them constantly with my GameCube. But there is high chance you bought knockoff SanDisk cards and they won't work.

Happy new year 🎉

sicounet, 2024-01-11:
Hello there . Just received my kingstone sd ... it worked immediatelly . On all consoles. Now i am so confused about those sandisks sd from ali . It is the first time i have issues with , and all on a single batch . For info the kingstone also comes from china market .

erlend0, 2024-02-17:
This should be in the troubleshoot section. I had the exact same issue. Installed Picoboot on two GCN and got the exact same issue trying multiple micro sd cards from Sandisk, opening the GCN redoing the wires, and yes even crying. Simply using Kingstone fixed it.