webhdx/PicoBoot #110: Invalid entry Point Issue when following the Compiling PicoBoot firmware Guide

Dracrius, 2024-05-01:
| Answer some general questions first                             	  |  	     |
|:-------------------------------------------------------------------	|:------:|
| Which firmware version are you using?                             	| `v0.2` |
Most recent main branch.
| Are you using the official Raspberry Pi Pico board?                 | yes/no |
Yes

When following 'Compiling PicoBoot firmware' (https://support.webhdx.dev/gc/picoboot/compiling-firmware) I am faced with an error about an Invalid entry point and base address 
```
~/Documents/PicoBoot$ ./process_ipl.py cubeboot.dol ipl.h
Entry point:   0x80003100
Load address:  0x00003100
Image size:    463264 bytes (452K)
Invalid entry point and base address (must be 0x81300000)
```

Unfortunately the guide doesn't explain what would cause this. I am attempting to make a build with the most recent changes and this fix for cubeboot (https://github.com/OffBroadway/cubeboot/pull/40). I've pulled @kemurphy branch for this pull request since @OffBroadway seems to be a little mia. I'd like to combine this version of cubeboot with the current picoboot so I have no questions of which version of picoboot I'm on.

I'll attach the builds I made. I'll be honest I have no idea what the ipl version is vs the cubeboot.dol but both folders create .dol files when you run make so I've tried using both with the pico boot processing script but both gave the same error with just slightly different addresses. I can only assume the ipl.dol is only the ipl while the cubeboot.dol is the intended dol.

[cubeboot.zip](https://github.com/webhdx/PicoBoot/files/15176256/cubeboot.zip)
[ipl.zip](https://github.com/webhdx/PicoBoot/files/15176257/ipl.zip)

webhdx, 2024-05-01:
The DOL file has to use the same entrypoint address as the original IPL (0x81300000). Cubeboot DOL is compiled to use a different address which makes it incompatible with PicoBoot. They have provided precompiled PicoBoot firmware in the past but I don't the details what kind of process they used to compile it. You have to try asking them for help.

Dracrius, 2024-05-01:
> The DOL file has to use the same entrypoint address as the original IPL (0x81300000). Cubeboot DOL is compiled to use a different address which makes it incompatible with PicoBoot. They have [...]

Thank you so much for the reply this wasn't what I was hoping to hear but fair. Unfortunately since said pull request has existed since February and is still yet to be accepted we don't seem to be getting much response from them. I'll try my best to track back when they created the precompiled version to see if I can get any hints on how they did it but it might be lost info.

Dracrius, 2024-05-06:
@webhdx I found a branch of cubeboot setup for Docker which saved me some hassle and showed I was still using the wrong .dol file from Cubeboot (I have no idea why it has so many with so little documentation on their uses) as it turns out it also has a folder called 'entry' that will generate one and this one appears to pass the first step
```

Entry point:   0x81300000
Load address:  0x01300000
Image size:    242632 bytes (236K)
Payload needs to be aligned to 237K
Output binary size: 242688 bytes (237.0K)
```

[entry.zip](https://github.com/webhdx/PicoBoot/files/15226516/entry.zip)

but unfortunately I've hit another issue that hopefully should be much easier to help with.

```
make
[  2%] Built target bs2_default
[  4%] Built target bs2_default_padded_checksummed_asm
[  5%] Performing build step for 'PioasmBuild'
[100%] Built target pioasm
[  6%] No install step for 'PioasmBuild'
[  7%] Completed 'PioasmBuild'
[ 12%] Built target PioasmBuild
[ 13%] Built target picoboot_picoboot_pio_h
[ 15%] Performing build step for 'ELF2UF2Build'
[100%] Built target elf2uf2
[ 16%] No install step for 'ELF2UF2Build'
[ 17%] Completed 'ELF2UF2Build'
[ 22%] Built target ELF2UF2Build
[ 23%] Building C object CMakeFiles/picoboot.dir/src/picoboot.c.obj
[ 24%] Linking CXX executable picoboot.elf
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: cannot find -lstdc++: No such file or directory
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/picoboot.dir/build.make:1207: picoboot.elf] Error 1
make[1]: *** [CMakeFiles/Makefile2:1511: CMakeFiles/picoboot.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

I've tried to track down the library I'm missing but I'm having absolutely no luck other then knowing it is a c++ library. If you could point me to the write command for linux that would be ideal as I have had so many issues trying to build on windows that I've been doing all my Gamecube / devkitpro builds through a Ubuntu 22.04 VM.

Dracrius, 2024-05-06:
Never mind I finally got a build made all of PicoBoot's build process works flawlessly on my windows setup so I was able to use Ubuntu to build cubeboot and Windows to build PicoBoot thanks for your help and amazing work! I look forward to the future as I see parts of the road map will eliminate most reasons to do this!