webhdx/PicoBoot #112: CLI file not found

deneraraujo, 2024-06-26:
| Answer some general questions first                             	  |  	     |
|:-------------------------------------------------------------------	|:------:|
| Which firmware version are you using?                             	| `v0.3` |
| Are you using the official Raspberry Pi Pico board?                 | yes |

After updated from v0.2 to v0.3, I got the message "CLI file not found... (release reset button to continue)" everytime I boot the console. 

[image]

Cionsole: GameCube DOL-001 Japanese

Mods:
- Pluto-IIx HDMI for GCVideo mod (https://www.knjn.com/ShopBoards_Pluto.html, https://www.youtube.com/watch?v=o0VOgJk0JgA&t=916s)
- Laserbear GameCube Blue Retro Internal Adapter (https://www.laserbear.net/products/gamecube-blue-retro-internal-adapter)

Tested PicoBoot versions:
- v0.1 - Swiss boot OK ✅
- v0.2 - Swiss boot OK ✅
- v0.3-beta1 - "CLI file not found" message ❌
- v03 - "CLI file not found" message ❌

webhdx, 2024-06-28:
Are you sure you're getting this error only on 0.3 releases? It feels weird because I don't recall any major changes in libogc or iplboot that would break compatibility with SD cards. 

I'll try to make a new release with latest libogc and gekkoboot over the weekend. Maybe that will fix the issue for you.

deneraraujo, 2024-07-01:
Thanks for answering.
Yes, I'm sure it's only on 0.3 releases.
Searching the Internet for this error message, I found something about sd card format problems. So I tried two other sd cards, but the message kept appearing. Only when I rolled back to 0.2 release, the message disappeared. So I updated again to version 0.3 and the message returned.

lopertyur, 2025-02-11:
I have a US GameCube DOL001 installed with version v0.3, wired with the scheme for that version. I run games from an SD SD2SP2 1.2a with a SanDisk Extreme 64 GB microSD. Everything works well, but when I turn on the console, that message briefly appears, similar to the one in the picture above, but without the last line that says 'Release reset button to continue.' It always shows up for a second every time I turn it on, and then it enters Swiss, and I can play normally. But why does that message appear?