webhdx/PicoBoot #114: [Question] Bridging pin1 to GND

Anakiev2, 2024-08-04:
| Answer some general questions first                             	|  	|
|:-------------------------------------------------------------------	|:------:	|
| Are you my GitHub sponsor? | no 	|
| Did you check older issues to see if it was requested before? | yes 	|

Hi and thank you for sharing this awesome project!

What is the purpose of this pin shorted to GND? I see people on youtube installing with the official guide and others installing with QSBs or flex cables who don't bridge the pin to GND but it's working.

What will happen if it's not bridged?
Shouldn't the the people with QSBs or flex cables also bridge the pin?

[image]

webhdx, 2024-08-04:
Serial Port 1 (the connector you bridge pin #1 to GND on) is sharing EXI bus with the IPL chip. However by default if no SP1 accessory is connected, data lines are not present on SP1 connector. Bridging pin #1 to GND tricks GameCube into thinking there is SP1 accessory and thanks to that we can steal some data pins from SP1 connector instead of having them soldered onto the IPL chip. It's actually safer and easier this way. Many people destroy IPL chip because it's small and fragile. Having to solder even more wires to it, would cause more failed installations.

Hope I explains it well.

Anakiev2, 2024-08-04:
Thanks for the fast response. I think I understand it.
In case of flex cable soldered directly on the IPL chip we don't steal data from SP1. So bridging is not needed.

webhdx, 2024-08-04:
Exactly 💯

closed 2024-08-05 by webhdx