webhdx/PicoBoot #120: GC doesn't boot to Swiss with pico firmware v0.4

jimfrob, 2024-08-11:
| Answer some general questions first                             	  |  	     |
|:-------------------------------------------------------------------	|:------:|
| Which firmware version are you using?                             	| `v0.4` |
| Are you using the official Raspberry Pi Pico board?                 | yes |

I have been using v0.2 for quite a while, but remembering that it was considered "highly experimental," I was keeping an eye out for updates. This morning's release of v0.4 didn't include the word "experimental," so I downloaded it, opened up my GC, and reflashed the pico. To my dismay, I got absolutely no output for the first several power-ons. When it did boot (which was extremely infrequent), it was to the regular GC system. Thinking my soldering might have come undone, I went to the soldering iron and redid the one connection on the pico that looked a bit dodgy. This made no difference. I redownloaded v0.2 and reflashed the pico with that. The GC booted right up to Swiss with that version. I deleted my download of v0.4 and redownloaded it and reflashed the pico again - again, the GC didn't boot. As per the notes in the release, I have not redone the wiring to remove the bridge between GP6 & GP7. Doing so would be difficult since I would have try to remove the pico. 

While shooting video for this issue, I did manage to get the GC to boot on v0.4 twice by fiddling with the angle of the wires, but more often this resulted in the console's booting into the regular system. I'm now starting to wonder if it might be the bridged GPIO. For now, I'm going to stick with v0.2. Even if it's experimental, it does work with all games I've tried playing.

In the attached videos, IMG_3550.mov shows a typical startup with v0.4. IMG_3551.mov shows a typical startup with v0.2.

https://github.com/user-attachments/assets/ada7e593-a6b2-4ec8-ac90-3347454fd35c

https://github.com/user-attachments/assets/4ad4e84c-8670-4cfa-8562-87513d3b407d

webhdx, 2024-08-11:
First thing I notice is that it would really help to shorten the wires. Unfortunately it's the most common issue people have with their setups. I know this Pico placement is suggested in Macho Nacho's video but much better location is near fan assembly as it allows for shorter wires run.

Can you check if this build helps to mitigate the issue? https://github.com/webhdx/PicoBoot/actions/runs/10342489277/artifacts/1799614245

It adds another safety measure to disable GP7 which could potentially degrade data signal. I haven't ran into this problem with 3 GameCubes I've tried with v0.4 though.

jimfrob, 2024-08-11:
You’re absolutely right about my following Tito’s guide on installing the pico. I didn’t know the wire length was going to be such a problem down the road. The build you linked did not fix the issue, so it probably is the wire length. I don’t feel like taking it all to pieces again to shorten them, so I’ll stick with v0.2. Maybe I’ll feel like it later. Thank you for your quick reply. I really appreciate this project. It allows me to play off of ISOs of my damaged discs (and maybe a few others 😉).

dragoncity17, 2024-08-12:
Same problem, 0.3 boot but if I flash 0.4 no boot.

So come back to 0.3 without problem.

webhdx, 2024-08-12:
@dragoncity17 Are you able to check if unbridging GP7 solves the issue for you? This is weird because GP7 should not mess the ouput, I've tested it on 3 of my consoles. But who knows...

dragoncity17, 2024-08-12:
I unbridged GP7 (only bridge GP6) same problem.

And of course now doesn't work with 0.3.

nextrizon, 2024-08-13:
I just flash v0.4 on an official Pico Pi with unbridged GP7 and 3.3V for power on a DOL-001. It will not boot Swiss(IPL.dol), just regular Cubeboot. When I bridge GP7, it will boot into Swiss with either SD2SP2 or Gecko SD. But when I remove SD2SP2/Gecko SD, Picoboot show status saying .DOL not found instead of booting Cubeboot.

I flash ith with v0.3 and bridge GP7, it will boot into Swiss, if there's a SD2SP2 with microSD card and regular Cubeboot when there is no SD2SP2.

Also I done multiple Picoboot installs, so I made the shortest wire length as possible and never had issue with my previous installs with v0.2 and v0.3 firmware.

Anakiev2, 2024-08-27:
This can be caused by cold solder joint on GP6 and a good one on GP7. I would reflow all pins on the Raspberry Pi Pico board or at least GP6.

walczakp, 2024-12-07:
TL;DR - check power rail soldering, this also seems to be important, v0.3 seemed to be less prone to issues with that.

Sharing my story to hopefully add some useful info on this issue. DOL-001 here, Japanese import, if that changes anything. Using original Wii power brick with modified plug. On my first attempt, using random bodge wires and 3.3V for quick test setup - v0.4 booted straight to Swiss with no issues. After ordering 1N5822 diode and some silicon 28AWG wire, I did it properly - 5V power with diode, almost exactly 10cm signal and gnd wires and v0.4 stopped booting (goes directly to the original GC menu). No GP7 bridge this time.

Then I did what's in the installation guide and reverted back to v0.3 while adding GP7 bridge and I'm booting to Swiss again. I also flashed v0.4 with bridged GP7 and it doesn't work.

Now I went to test 3.3V again, so I removed 5V wire from GC, soldered a quick random length bodge to 3.3V and v0.4 started booting again. Even better - resoldering 5V again now brought me back to bootable v0.4 (I still kept GP7 bridged though). Even though I was sure I did it right the first time, now it apparently runs fine. It could have been something else as well, I have no way to tell now. I'll report if it becomes unstable somehow.

TorqueEmPup, 2025-01-20:
Can confirm 0.4 fails to boot on my setup. 0.3 boots perfectly fine.
Yes my wires are a little long, but imo if things work perfectly on one version, they should in the next.

Shortening wires because of a update to the software imo is not an acceptable solution.