webhdx/PicoBoot #122: Panasonic Q with original wiring method boots to GC Menu on v0.4

Dlang6487, 2024-08-19:
| Answer some general questions first                             	  |  	     |
|:-------------------------------------------------------------------	|:------:|
| Which firmware version are you using?                             	| `v0.4` |
| Are you using the official Raspberry Pi Pico board?                 | yes|

Picoboot v0.4 does not boot Swiss v.1695 using a SD2SP2 1.0 and 2.0 adapters from Blackdog Tech and from Laserbear respectfully. It boots to the Gamecube menu. Wiring is the original method, and flashing the Pico back to v0.3 or v0.2 boots to Swiss instantly.

webhdx, 2024-08-19:
A few people reported this issue. I'm currently looking for a solution.

Anakiev2, 2024-08-27:
This can be caused by cold solder joint on GP6 and a good one on GP7. I would reflow all pins on the Raspberry Pi Pico board or at least GP6.