webhdx/PicoBoot #123: Support for PICO 2

SmakyTaky, 2024-10-13:
| Answer some general questions first                             	|  	|
|:-------------------------------------------------------------------	|:------:	|
| Are you my GitHub sponsor? | no 	|
| Did you check older issues to see if it was requested before? | yes |

Will Raspberry Pico 2 be supported? I tried to complile the source code with both Visual Studio Core + Pico Extension and from command line (after installing cmake and phyton 3. Both would not complete complilation and show errors. Albeit I am a newbie when it comes to PICO compiling, I was not able to figure it out.

thericeking, 2024-12-21:
Can we expect Pico 2 support sometime in 2025?
PS. Thank you for your work and Merry Christmas.

webhdx, 2024-12-21:
Yes I already have the firmware ready, just need to test it. I hope to release it around Christmas.

SmakyTaky, 2024-12-23:
These are great news!

VersatileNinja, 2025-03-04:
Checking to see if any plans to release pico 2 support this year?

webhdx, 2025-03-04:
Yes I plan to work on it shortly.

webhdx, 2025-06-05:
Check the latest release [v0.5.0](https://github.com/webhdx/PicoBoot/releases/tag/v0.5.0). Pico 2 support was added.

closed 2025-07-19 by webhdx