webhdx/PicoBoot #124: Issue Launching Swiss with HDMI Adapter Connected

LazarusQ, 2024-12-19:
| Answer some general questions first                             	  |  	     |
|:-------------------------------------------------------------------	|:------:|
| Which firmware version are you using?                             	| `v0.4` |
| Are you using the official Raspberry Pi Pico board?                 | no |

Hi, so I've recently modded one of my GameCubes and successfully installed a Pico board using a flex cable. The issue I'm running into, however, is that when I try to boot the GameCube up with a HDMI Adapter connected it boots straight into the generic GameCube boot menu and the LED on the Pico board doesn't light up, but if I disconnect the HDMI Adapter first, power on the GameCube and wait for the LED to turn green on the Pico board before reconnecting the HDMI, it boots successfully into Swiss and I have no issues thereafter. I'm wondering if perhaps this is a power issue and the HDMI is taking priority over the Pico board? Perhaps if I solder the VSYS pin on the Pico and flex cable instead of 3V3 it might fix this issue by supplying enough power for both the board and HDMI Adapter to run together on launch? Any help would be greatly appreciated, I just want to make sure before I resort to trial and error and potentially brick my console.

webhdx, 2024-12-19:
What kind of HDMI adapter do you have?

LazarusQ, 2024-12-19:
Thank you for the fast reply! I'm currently using the Bitfunx NGC HDMI Adapter.

webhdx, 2024-12-19:
This is a hardware issue with the HDMI adapter. To mitigate the problem you should power Pico from 5V but that would be impossible with a flex cable.

LazarusQ, 2024-12-19:
Ah, I see. Are there any alternative adapters that you know of that would work with the 3.3V?

roadkillusa, 2024-12-19:
I have used this HDMI adapter with no issue on 3.3-volt Pico boot hookup
https://www.amazon.com/gp/product/B083RXFNDD/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

evilgenius31, 2024-12-19:
The 3d printed mini-hdmi adapters on aliexpress work great with picoboot!

LazarusQ, 2024-12-20:
Thank you for the replies everyone! Has anyone here used the Retro+ HDMI Adapter from Blue Retro and know if it works with 3.3V Pico on startup?

Ztokke, 2025-10-28:
So I just finished my installation, also with a ribbon-cable (using genuine Raspberry Pi Pico) and using the Carby Insurrection Adapter (with the latest firmware). Swiss Boots perfectly the first time, but then there is no video after power cycle. I read in some other issue that this was thought to be limited to other adapters but I can confirm this does occur on the Carby ones too.

If I connect it to a normal scart connector I get video signal and boots into swiss perfectly each time.

I was going to look into the 5V solution but say your comment that this is not possible with the ribbon cable sadly.

Is there a software solution possible for this or should I desolder everything and go for cables instead?