webhdx/PicoBoot #125: Clean Boot

lopertyur, 2025-02-12:
| Answer some general questions first                             	|  	|
|:-------------------------------------------------------------------	|:------:	|
| Are you my GitHub sponsor? | no 	|
| Did you check older issues to see if it was requested before? | yes 	|

Is there a way to prevent text from appearing when the console starts? That text that says 'Searching M.2, USB Gecko, sd2,' etc.

webhdx, 2025-02-19:
It's currently not possible but definitely a great idea for a feature.

lopertyur, 2025-02-20:
> It's currently not possible but definitely a great idea for a feature.

Thank you for responding, I take the opportunity to see if you can help me with this question:

I have a US GameCube DOL001 installed with version v0.3, wired with the scheme for that version. I run games from an SD2SP2 1.2a with a SanDisk Extreme 64 GB microSD. Everything works well, but when I turn on the console, this message briefly appears:

[image]

and then it enters Swiss, and I can play normally. But why does that message appear? and why says "CLI file not found, unmounting sd2" but enter to swiss normally?