webhdx/PicoBoot #130: Can no longer boot to OEM Nintendo menu by default anymore.

Deathstryker, 2025-06-04:
| Answer some general questions first                             	  |  	     |
|:-------------------------------------------------------------------	|:------:|
| Which firmware version are you using?                             	| `v0.5` |
| Are you using the official Raspberry Pi Pico board?                 | yes |

Hello all, I installed Picoboot when it first came out and it worked great. I had boot set up so that by default it launches the OEM Nintendo menu and only by holding buttons do you go into the various Apps like Swiss. I updated to 0.5 today and found that this boot method doesn't seem to work. It hangs up looking for ipl.dol and I never used ipl.dol because I never wanted it to launch an app by default. What can I do to solve this?

JesusSanchezLopez, 2025-07-23:
I am just adding that I am experiencing the same issue. Went from v0.2 to v0.5 and it no longer boots to standard IPL by default.

webhdx, 2025-07-23:
No, it was changed in gekkoboot. The reason behind that change was that many people couldn't tell PicoBoot was working because it was booting into stock System Menu without any visible information.