webhdx/PicoBoot #133: No DOL Loaded

Vzolfo24, 2025-07-01:
Trying to get Picoboot up and running on a DOL-101 console, but it refuses to boot into Swiss and only boots into the normal GameCube BIOS. I'm using Version 2.0 for Picoboot, and I do see the debugging text when I hold down on the d-pad when I boot up the console, to see that it says No DOL Loaded. I verified that the wiring is correct without any shorts, tried multiple different Pico chips, a hardwired SD to microSD card adapter soldered to Mem card slot B, and an SDGecko to no avail. I have done 30+ installs at this point and haven't come across this issue before. The microSD card is formatted to FAT32 with a 32k cluster size, and the only file on it is Swiss renamed to IPL.dol. I'm using known-working microSD cards, SDGecko, and a Pico chip. Any assistance would be greatly appreciated.

Vzolfo24, 2025-07-01:
It still says "No DOL Loaded" even with the swiss.dol renamed to ipl.dol

webhdx, 2025-07-02:
Can you tell more about your setup? I would try with a different SD adapter and card.

Vzolfo24, 2025-07-02:
Yes, and it still doesn't boot into Swiss and says that there is no DOL loaded