webhdx/PicoBoot #137: Sometimes clean boot, sometimes it goes to Swiss

Giankaolo, 2025-08-06:
| Answer some general questions first                             	  |  	     |
|:-------------------------------------------------------------------	|:------:|
| Which firmware version are you using?                             	| `PicoBoot v0.5.0 - picoboot_full_pico.uf2` |
| Are you using the official Raspberry Pi Pico board?                 | yes, with type C usb |

Hi, yesterday I tried to mod my GameCube (I believe it's a DOL-001) using a Raspberry Pico (v1) with a USB-C port and an SD2SP2. I struggled a lot with the soldering, but in the end, I connected the wires (small ones I found in the garage, not sure about the gauge). Right after soldering the Pico, I turned on the console and saw the error screen, so everything seemed fine.

Then I reassembled the console, and from that moment on, things got weird: sometimes it boots into Swiss, other times I get the Nintendo screen to select the language (clean boot). If I turn the console off and on 10 times, maybe 1-2 times it boots into Swiss and the other 8 times it does a clean boot.

By the way, I'm using a 64GB Kingston SD card that I partitioned down to 32GB and formatted to FAT32.

Any ideas on what I should do?

Giankaolo, 2025-08-07:
> USB-C Picos are known to have issues with PicoBoot. They use low quality components and sometimes they cannot withstand the overclock.
> 
> You should check your cable length. PicoBoot is very sensitive to the type and length of the wire. Please refer to the documentation: https://support.webhdx.dev/gc/picoboot/installation-guide
> 
> Try booting it while holding Reset button on the console. If you get reliable boot then the problem lies in the SD card setup.

I tried booting the console while holding the Reset button. When I do that, I get a black screen that says:
"gekkoboot r9.4 (with M.2 Loader support) Skipped. Rebooting into original IPL... (release reset button to continue)".

> Why would you partition your SD card? It isn't something I recommend.

Because the SD card I have is 64 GB (Kingston Canvas Select Plus), and I had read that it needed to be formatted in FAT32. But now I’ve seen in the guide you linked that exFAT should work too, so maybe I can just proceed like this and try again.

webhdx, 2025-08-07:
Test it with Reset button a few times. If you always get it to gekkoboot screen then the problem is your SD card. 
You can format 64GB card to FAT32. FAT32 is more compatible with different homebrew but if you only care about Swiss then exFAT is fine.