webhdx/PicoBoot #139: Boots to normal GameCube menu, acts like it did from the factory

Rainbow2600, 2025-08-24:
| Answer some general questions first                             	  |  	     |
|:-------------------------------------------------------------------	|:------:|
| Which firmware version are you using?                             	| `v0.3.1` |
| Are you using the official Raspberry Pi Pico board?                 | yes |

Japanese NTSC-J DOL-001 (Spice Orange)

I have tried the newest version of PicoBoot, which did not work. Booted to stock menu.

Downgraded to 0.3.1v per Github just incase, and bridged GP7. Installed a new Pico as well because why not.

Both times uf2 caused the device to eject, properly

So two versions, two Picos. Still nothing.

I shortened my 26 AWG wires to almost too short. Still nothing.

I completely redid all my solder joints even though I was sure they were good the first time, and even cut and restripped the wires for a nice clean piece to tin freshly.

I did read that 0.3.1v will boot to the default menu unlike newer versions of the SD card is amiss, which in my case I haven't even bothered to mess with, as I can hold reset while powering on to test the PicoBoot install itself

It should do this test even without an adapter & card connected, yes? I'm going to look really stupid if my method of testing is what is causing the problem since I have stopped connecting the SD card + memory card adapter to try and ensure I have the hardware side working first before I proceed and add more failure points

My lens is busted on my phone so I can't provide great soldering pictures, but I can show wire length. (I also neglected to take pictures of my soldering job...)

It should be noted I soldered on the reverse of the Pico and in my infinite wisdom I took a picture of the wrong side, but those joints are good. I haven't gotten the chance to open the GameCube again yet to take more pictures.

I formerly soldered for a living for many many years and have $800 of Hakko equipment just in my home workbench. This is a bit smaller than what I am familiar with working on (tiny through-hole components) but I know how to lay a good solder joint nonetheless. Thanks for any help!

The console still works perfect and I was really particular about not bridging any pins or traces (I assume it would exhibit issues rather than just not work, if I had)

Cool project just wish I could try it! :P

[image]

webhdx, 2025-08-24:
It's better to test by holding RESET button on the console. It always behaves the same on all gekkobbot versions and it doesn't require SD card so it's a great method to validate your installation.

Do you get a green light on Pico?

It would be great if you could provide photos of the actual solder joints on the motherboard.

Rainbow2600, 2025-08-25:
I soldered the wires to the correct place and it worked first time with the reset test. Sorry for wasting your time. Really can't believe it! Maybe my blunder will help somebody some day that also had not enough sleep before trying this out. Or a few beers...

Cheers and thanks! Excited to have PicoBoot working!

closed 2025-08-25 by Rainbow2600